AmberELEC/AmberELEC #218: 2.0 will not revognize device as a ps1 controller

FB74, 2021-03-01:
### Describe the bug
Several ps1 Games/roms starts up and works fine in 1.11 but when using 2.0 they only starts up. When the intro s are over the device is not recognized as a ps1 controller. The roms are in the right folder what I can see. Same bios as in 1.11. 
File extension problem? Shouldn't be because works fine in 1.11?

Regards 

### To Reproduce
Steps to reproduce the behavior:
Start a ps1 game. 

### Expected behavior you can't play the ps1 games. Leave content only option. 

### the games says put in a ps1 controller. 

### rg351p 351elec 2.0

fewtarius, 2021-03-01:
The solution to configuring the various PSX controller modes can be found in the WIKI.

closed 2021-03-01 by fewtarius

FB74, 2021-03-01:
Do you really need to confiq the rg351p device as a controller?

fewtarius, 2021-03-01:
> Do you really need to confiq the rg351p device as a controller?

https://github.com/351ELEC/351ELEC/wiki/Getting-to-Know-351ELEC#playstation-default-controls