AmberELEC/AmberELEC #220: Request for a BBC Micro emulator

SuperBadger77, 2021-03-02:
Please could I request the addition of a BBC Micro emulator. It's my favourite system (along with the Acorn Electron) of all time. I lot of good memories and I bet others would appreciate it. It's available on The RA, and via Advance MESS.

A port was recently made for the RG350 which can be found here:
https://github.com/jeremyrayner/beebem-rg350 the dev made some changes to record the key input to joypad mappings saved per each game.

Original source code here: http://beebem-unix.bbcmicro.com/download.html

A bunch of git resources here: https://github.com/stardot

Original source used for the RG350: https://github.com/thefossilrecord/beebem-rs97

Would be greatly appreciated.

stickfight, 2021-03-03:
oooh original elite!

SuperBadger77, 2021-03-03:
Yes - in all it's glory :-) does this mean there is a chance? :-)

SuperBadger77, 2021-03-04:
👍🏻😃

fewtarius, 2021-03-06:
We're not going to add this one, but if someone in the community wants to contribute the package we'll gladly merge it in.

closed 2021-03-06 by fewtarius

ShamelessPolymorph, 2021-03-06:
Fascists

SuperBadger77, 2021-03-06:
> We're not going to add this one, but if someone in the community wants to contribute the package we'll gladly merge it in.

Can you take the source code from the links above, compile it, and merge that package in?