AmberELEC/AmberELEC #221: Poor Visual Performance in OpenTyrian Port

jairova, 2021-03-03:
Running version 2.0 (5e6b2b7) on the Stable Channel, on a RG351P without internal wifi
**Describe the Bug**
While playing OpenTyrian, in any of the levels, the framerate is noticeably choppy, and there is some screen tearing.  Input response seems unaffected.
**Steps to Reproduce**
Start the Tyrian port.

fewtarius, 2021-03-03:
Would you mind trying the nightly build and let me know if it corrects it?  There are only minor changes there so it should be as stable as the build you're on now.

jairova, 2021-03-03:
> Would you mind trying the nightly build and let me know if it corrects it? There are only minor changes there so it should be as stable as the build you're on now.

Just updated.  No dice unfortunately.  I can say though that the framerate, though low, doesn't fluctuate much, at least it's consistently choppy throughout the level, number of sprites/particles doesn't to affect game speed.

fewtarius, 2021-03-03:
Oh, you should switch back to the stable channel if you haven't yet or you'll be prompted to update over and over again.  The nightly and RC channels are configured that way for testing.

Tokodomo, 2021-05-31:
Same issue as: https://github.com/351ELEC/351ELEC/issues/390#issuecomment-851584342: „_The reason is a faulty or improper build/version of SDL (e.g. libSDL2-2.0.so)..._“

closed 2021-05-31 by dhwz