AmberELEC/AmberELEC #224: REQUEST: OneShot port

thisjustin2001, 2021-03-04:
Hey! As the title says, I'm requesting a port for the freeware game OneShot to 351Elec! OneShot is a freeware RPG Maker game (not to be confused with the paid remake that uses a later RPG Maker), and EasyRPG is already on 351Elec via RetroArch. But OneShot controls the runtime externally, which other RPG Maker 2000/2003 games do not do, so the EasyRPG player is (understandably) incapable of running OneShot. Luckily, there were people dedicated enough and desperate enough to play OneShot that they made a specific fork of EasyRPG built specifically to run OneShot!

So, what am I asking you to do? If you'd be willing, I'd be thrilled if you'd be willing to port the OneShot fork of EasyRPG to 351Elec, so that fans of OneShot, or first time players can experience the game on the RG351 using 351Elec! Here's the link to the fork of EasyRPG: https://github.com/20kdc/EasyRPG-Player-Fork-For-OneShot-Legacy

Oh, one last thing! If you just port the fork of EasyRPG, users maybe could add the necessary game files for OneShot themselves. Maybe you or me could commission Retro Game Corps to make a guide on it if it ends up being a confusing process!

Thank you so much for reading my request! Whether or not you end up porting this for me, I'm still so grateful for everything you've done for the RG351 community. Of course I'd be more happy if I could play OneShot on my 351, but still... You're the best! Thank you!

fewtarius, 2021-03-06:
We don't really focus on ports, but if you ask in the ports channel on our discord maybe one of our community members will take it on.

closed 2021-03-06 by fewtarius

thisjustin2001, 2021-03-07:
Hey! I really appreciate that idea and the Discord invite! I think I'll do that! Trouble is, there's no where in the Discord where I'm allowed to say anything? I'm a complete Discord newbie, so maybe I'm doing something wrong? I know you probably have better things to do then help someone learn Discord, but maybe you could tell me what I'm doing wrong?

thisjustin2001, 2021-03-07:
I figured it out! I reacted to the rules post again with a generic thumbs up, instead of the toon Link thumbs up! Sorry for bothering you with two unnecessary emails!