AmberELEC/AmberELEC #227: Sleep mode not working as expected when setting screensaver to sleep

xserk, 2021-03-05:
### Describe the bug
Coming out of sleep mode doesn't work as expected when you have set "sleep" as screensaver option.

### To Reproduce

1. Set screensaver time to 1 min.
2. Select sleep as screensaver.
3. Put device to sleep pressing the power button.
4. Wait for 1 min.
5. Press power button to bring the device out of sleep mode.

### Expected behavior
The device's screen should turn on. Instead it turns on and then turns off again.

### Device (please complete the following information):
 - Version 2.0

### Additional context
The bug is not reproduced if you don't wait for the ammount of time you have set at the screensaver before you try to bring the device out of sleep.

fewtarius, 2021-03-05:
Thanks for the report, this is a known issue but we don't have a fix for it yet.

xserk, 2021-03-05:
No problem, thank you for making such usable software.

fewtarius, 2021-03-06:
Should be fixed by https://github.com/351ELEC/351elec-emulationstation/commit/4ed9bf0b9d56511b754a67ef4748508f2767b244

closed 2021-03-06 by fewtarius

xserk, 2021-03-07:
Is the fix available on RC or Nightly?