AmberELEC/AmberELEC #231: Multiple CDs games showing many times on EmuStation interface

chfreis, 2021-03-07:
Having solved this problem on PC enviro by manipulating "**es_system.cfg"** and changing the extensions to show on the front-end, I thought the process could work analogous on the device.
So there i went hunting for that file on the partitions created by 351ELEC, found it, changed it, but to my surprise the OS identified my changes and refused to boot emustation, showing an error message that configs were tempered with.
And my feeling (i did not test this) is that even if i had backed-up the files before hand, the OS would have identified the changes on the **"storage"** partition and not let me boot it anymore, because it looks like there are hard-constraints in place for changing files on that specific partition (correct me if I'm wrong).

I don't know if that's working as intended and to my knowledge there's no other way of doing this process (to not show multiple discs of the same game as "different games" multiple times) by configuring emustation via device's UI configuration.

I think you guys should either implement UI changes to let us choose what extensions we want to list as games in emustation UI or at least not have the hard-constraint of not being allowed to temper with configuration files on **"storage"** partition.

I thank you all for your time and admirable work on creating a much better and friendly OS for this device.

anthonycaccese, 2021-03-07:
There is a way directly in ES to hide/show specific file extensions on a per system basis.  

On a given system press "select"
In the menu that opens choose "view customization"
Then choose "file extensions"

In that menu you can choose which extensions to hide/show by unchecking/checking each

chfreis, 2021-03-07:
> 
> 
> There is a way directly in ES to hide/show specific file extensions on a per system basis.
> 
> On a given system press "select"
> In the menu that opens choose "view customization"
> Then choose "file extensions"
> 
> In that menu you can choose which extensions to hide/show by unchecking/checking each

I apologize for my ignorance.
Thank you, kind sir.

closed 2021-03-07 by chfreis