AmberELEC/AmberELEC #234: Mapped right analog stick to L2/R2, now will not unbind.

kartsnkittiesGF, 2021-03-09:
**Describe the Bug**
PSX on pcsxe core. Playing Spyro 3. Decided to try to map L2/R2 to left/right on the analog stick, apparently there are 2 sections for remapping controls and I think I used the wrong one. Now in all my PS1 games, right analog stick left/right is registering as L2/R2, and I've tried resetting the controls to default, remapping them, etc. but even though they show as being assigned to the correct inputs, right analog stick is still controlling those buttons. L2/R2 still work as intended, and right analog left/right still work, but when I push it left or right it acts as though both the analog stick and trigger were being pressed. When i go to the correct mapping section and set L2/R2 to different buttons, the analog stick works as those buttons. So if I set L2 to X, the analog stick will work as X. Nothing I do is fixing this, please help
**Steps to Reproduce**
Decide to remap your analog sticks to buttons using the incorrect remapping interface. RIP
**Expected behavior**
In such a scenario, setting the affected buttons to proper values, or using the reset to default controls function, should at least do something..No value shows the analog stick as being bound to the triggers but for some reason no amount of setting or clearing those buttons changes that they are in fact bound to the analog stick permanently now
**Device**
Running version 2.0 (5e6b2b7) on the Stable Channel, on a RG351P without internal wifi

kartsnkittiesGF, 2021-03-10:
I'm in the Discord, and I don't see a tab that says Support/Troubleshooting. There are 8 tabs that I can see, Rules, Welcome, Announcements, Github, Releases-nightly, Releases-testing, Releases-stable, Survey-says, and all 8 of these tabs say 'You do not have permission to send messages in this channel.' Can you help me with my issue, or direct me to somewhere where I can be helped? I've yet to fix this issue.

fewtarius, 2021-03-10:
You have to read and accept the rules before the general channels open up.