AmberELEC/AmberELEC #237: custom launching images possible?

Manfred282, 2021-03-12:
Is it possible to add support für custom launching images, like these ones for retropie?

[https://github.com/lipebello/retrorama-launching-images](url)

There is a option under "Games Settings" called "Enable Game Splash", but it only enable/disable the Loading "gamename" splash.

Retropie has runcommand, would be nice to add a similar option ;)

[https://retropie.org.uk/forum/topic/4611/runcommand-system-splashscreens](url)

fewtarius, 2021-03-12:
It is not possible, no.  The built in game splash is the only available splash, and we aren't planning on adding support for custom images.

closed 2021-03-12 by fewtarius

jimmyff, 2021-05-12:
I think this is an awesome suggestion, plus it would make those 5-10 seconds whilst a rom loads feel better as sometimes I wonder if it crashed