AmberELEC/AmberELEC #239: new 351p wont launch, error message says SYSTEM FAILURE, check kernel and dtb images

jakemoore2, 2021-03-14:
like i said the brand new device wont boot, i bought a new tf chip and copied emuelec onto the card. the device meaning the console is not able to be discovered by my PC, any help?

DrorC, 2021-03-14:
Hi. same issue on my device. On my case after installation the 351elec ver. 2 on new SDcard by the instructions i got "system error" at the screen (kernel and dtb files.) by trying to bootup the device (tried several times) i even tried on 2 new SDcards. when inserting again the original SDcard there is no errors and runs fine.
can it be upgraded to 2.0 from the stock firmware or just installation from scratch?
Did Someone manage to install and can send link to image?
Thanks

jakemoore2, 2021-03-14:
This didnt fix the problem for me..

DrorC, 2021-03-15:
Helped for me. i extracted the .gz file with 7zip then i found that the file size is 1gb and not 600mb then flashed this extracted file and work. hope it will work for you also. good luck

jakemoore2, 2021-03-15:
I'm confused as to actually how and what to do.   Would you mind a phone
call with me?

bitkoin, 2021-03-15:
are you using balenaetcher or win32 disk imager to flashing firmware? Stop use and download rufus to flashing 351elec instead. U dont have to extract the compressed file

fewtarius, 2021-03-15:
Extract the image before writing it.  Instructions for how to flash the image are documented on the wiki.  If you need further assistance, drop by our discord.

closed 2021-03-15 by fewtarius

jakemoore2, 2021-03-16:
I'm actually confused as to what to do at all...... at this point I have
the original card, another tf card..I dont know how to flash firmware or
anything can u help me find instructions