AmberELEC/AmberELEC #243: RA Load Remap File doesnt work / crashes game

Manfred282, 2021-03-17:
### Describe the bug
Loading a Game Remap File wont work, it will instantly crash the game.
I tried with different cores/systems, behavior is the same.

### To Reproduce
Steps to reproduce the behavior:
1. open RA Menu -> Quick Menu -> Controls -> Port 1 Controls
2. change any key
3. backout and select "Save Game Remap File" 
4. load another game
5. open RA Menu -> Quick Menu -> Controls -> Load Remap File

### Device (please complete the following information):
351ELEC 2.0

dhwz, 2021-03-18:
Doesn't happen every time, but yes I had that also 2 times today.

bebeidon, 2021-03-19:
i think this only works when there are the exact same buttons available for these specific roms you load and apply to. but even when there are 3 buttons for both games there can be differences as it seems, as it still crashes sometimes when the number of button should be correct.

dhwz, 2021-03-19:
Yes maybe I didn't check if the mapping is 100% identical. I just tried to copy the remap for some vertical games in FBNeo, then it crashed but when I tried it again it didn't. But still its confusing it shouldn't crash even if the remap doesn't match for the selected game.

Manfred282, 2021-03-19:
> i think this only works when there are the exact same buttons available for these specific roms you load and apply to

Dont think so, i saved the mapping for Mortal Kombat 1 (MAME), loading it into MK2 or 3 make it crash.
As you can see, the mappings for SF2 and SF2CE are identical, but loading them still crashes the game.

[image]