AmberELEC/AmberELEC #244: New Theme - Nes Box

luciano2000, 2021-03-17:
This theme is awesome and its a default for ArkOs.

I love to see this theme as 351ELEC default too.

Here is a retropie URL: https://retropie.org.uk/forum/topic/27698/es-theme-new-nes-box-theme
end
here is a github url: https://github.com/KALEL1981/es-theme-nes-box

anthonycaccese, 2021-03-17:
We are not able to use the nes-box theme directly for 2 reasons:
1) it needs to be modified from its native aspect ratio of 16:9 to better fit the 3:2 aspect ratio of the rg351
And 
2) it’s license prohibits that type of modification (as seen on its GitHub repo)

Because of those 2 reasons we will not be able to bring this theme in with the needed modifications for aspect ratio into our theme downloader at this time.

closed 2021-03-17 by anthonycaccese

szalik-rg351, 2021-03-17:
There will be our version of nes box eventually in future :) Due licensing problem we can't use Kalel1981 work, but we can do something inspired of his work