AmberELEC/AmberELEC #333: Roadmap & Feature Wishes

Cebion, 2021-05-01:
## Base-OS
- [ ] Add MP3 BGM support
- [ ] Update RunEmu Scripts to reduce loading times
- [ ] Introduce Docker Container
- [ ] ~~Update support for bluetooth devices~~
- [ ] ~~Add network proxy support~~
- [ ] ~~Look to add rumble support to more emulators~~

## Emulators
- [ ] Update to native OpenBOR from Batocera. See: https://discord.com/channels/777665344289898536/777943158579134474/804201211473428532 & https://discord.com/channels/777665344289898536/795710647303536670/813523328437583904
- [ ] Research Saturn emulator not booting
- [ ] Add standalone Mupen64Plus emulator
- [In Progress] Evaluate adding standalone Dreamcast emulator from RetroRun (see: https://www.youtube.com/watch?v=vfL3ohWaYgk)
- [ ] Investigate improvements to overlay support
- [ ] Consolidate bbs and carts directory for Pico-8

## EmulationStation
- [ ] Come Clean with emulationstation-batocera
- [ ] Update the package tool to add package name and description
- [ ] Add TATE mode config to EmulationStation menu (upstream)
- [ ] Correct the retroachievements special character in password issue. Also check the screen scraper to see if there's an issue there as well.
- [ ] ~~Add moonlight config to EmulationStation menu~~

## Ports
- [In Progress] Explore adding Sonic 1,2,CD decompilation from EmuElec
- [ ] Update Solarus with gamepad patch
- [ ] Update to latest OpenBOR version

anthonycaccese, 2021-05-04:
For ES slowdown these look to be the 2 commits we can reference from EmuELEC - 
https://github.com/EmuELEC/emuelec-emulationstation/commit/01a0cdc37bbfd02bb5fbd807bc2bc238917f31ec
https://github.com/EmuELEC/EmuELEC/commit/d4f82d483889994ab040a49ad7f8c2f67b3befbb

dhwz, 2021-05-05:
> 
> 
> For ES slowdown these look to be the 2 commits we can reference from EmuELEC -
> [EmuELEC/emuelec-emulationstation@01a0cdc](https://github.com/EmuELEC/emuelec-emulationstation/commit/01a0cdc37bbfd02bb5fbd807bc2bc238917f31ec)
> [EmuELEC/EmuELEC@d4f82d4](https://github.com/EmuELEC/EmuELEC/commit/d4f82d483889994ab040a49ad7f8c2f67b3befbb)

I''ll review these changes if nobody is still working on that right now.

madmanRB, 2021-05-05:
Oh I hope that RetroRun comes to 351elec!

szalik-rg351, 2021-05-05:
i send greetings to my Grandmother, thanks Cebion

nadameu, 2021-05-05:
Is it possible to allow updating Retroarch cores directly from the libretro buildbot? I believe ArkOS does this.

Cebion, 2021-05-05:
> Is it possible to allow updating Retroarch cores directly from the libretro buildbot? I believe ArkOS does this.

Not really since we build the cores from source and they land in the read only partition.

takano32, 2021-05-06:
How about OTA update?

https://github.com/351ELEC/351ELEC/commit/219fb607a8e1ff17ec15c6e9ac0dddb837562efd