AmberELEC/AmberELEC #339: Solarus

sickhate, 2021-05-01:
im using a fresh install of 351elec. Nothing installed, except for solarus roms and it does not run any rom. I have used the same roms on other firmware and it runs very well. Could you please look into this?
Im using the final build with fresh clean install

szalik-rg351, 2021-05-02:
rg351v or p/m ?

sickhate, 2021-05-02:
Rg351v, i have tried a bunch of trouble shooting and i cant get it to work…it all works fine on any other firmware

I absolute adore this firmware

Marco Viveiros
& Registered Nurse
& Certified premier personal trainer
& Fitness instructor
& Nutritional Advisor
& Competitive Bodybuilder

Cebion, 2021-05-02:
> im using a fresh install of 351elec. Nothing installed, except for solarus roms and it does not run any rom. I have used the same roms on other firmware and it runs very well. Could you please look into this?
> Im using the final build with fresh clean install

Did you download the Solarus Package? There are no solarus bios files.
You download the package and put your games under roms/solarus

sickhate, 2021-05-02:
Yes i have placed the games under the right folder and i have tried the same roms on other firmware and they work but not on 351elec.

Cebion, 2021-05-02:
> Yes i have placed the games under the right folder and i have tried the same roms on other firmware and they work but not on 351elec.
> The wiki doesnt mention any package…is there any particular file i need to place in there?!

https://github.com/351ELEC/351ELEC/wiki/Supported-Emulators-and-Ports#community-built-packages
3rd entry solarus under Community built packages
> 
> ps i checked on solarus website i can only find windows, mac and rasperry packages…could i get some enlightment on what to do because on the wiki page i cannot find any info on this and i have googled everything and cannot find a single piece of information. Didnt get any help on discord either…

I didn't see any questions from you on discord, in what channel did you ask?
Normally we respond to each question. Please come visit us there so we can help you get it going.
> 
> ps: if u mean the package installer incluided on 351elec i dont have any option for solarus. Im only shown 2 option
In the package installer you should have a list of +15 entries. Make sure you're connected to the internet.

closed 2021-05-02 by Cebion