AmberELEC/AmberELEC #340: [Solarus] Buttons mapped incorrectly or not mapped.

LeBiquet, 2021-05-02:
I'm on the last version of 351ELEC .

Games load up fine and are beautiful.

In "Return of the Hylian" and "Mystery of Solarus DX", I can't move with the D-PAD. Joystick is mapped but mirrored (UP= DOWN etc.) and has insane dead zones.

sickhate, 2021-05-02:
U can changed them within the game. I had that issue also. Press menu go to options and change keymap. Worked very well for me. Save the file and thats it

Cebion, 2021-05-02:
This is sadly an issue on solarus that can't be fixed on our end.
They already have an issue open for it.

anthonycaccese, 2021-05-03:
Linking solarus issue for tracking purposes: https://gitlab.com/solarus-games/solarus/-/issues/1523

Closing for now on our end as we wait for a fix upstream

closed 2021-05-03 by anthonycaccese