AmberELEC/AmberELEC #353: EasyRPG games choppy and slow

fry-guy, 2021-05-06:
EasyRPG games currently play very choppy and slow. Basically unplayable. I'm specifically encountering this issue with Legion Saga and the Ara Fell Demo. Others have reported similar issues: https://www.reddit.com/r/RG351/comments/my8ix3/351elec_easyrpg_games_very_choppy/ 

Similar issues do not seem to exist in ArkOS, in my experience. Could this be addressed before your final update, please?

I am running firmware version 20210422. 

Thank you!

dhwz, 2021-05-06:
Ok we have to check if there is a different version used in ArkOS.

FYI there will be mo final update.

fry-guy, 2021-05-06:
If it's helpful, it appears my version of ArkOS is running 0.6.1 6ad57329, whereas 351Elec is running 0.6.1.

My ArkOS installation is also on EmulationStation 1.9.0, whereas 351Elec says 1.9.1.

dhwz, 2021-05-06:
hm I can't find the hash 6ad57329 on the easyrpg git, strange

fry-guy, 2021-05-06:
It's with all the games I tested. Which isn't many, admittedly. Just tried Villnoire, though, and it's also super choppy.

Here's a screenshot of the RetroArch core information page for EasyRPG on ArkOS: https://imgur.com/gallery/1zbSTGA

I also checked the ~/home/ark/.config/retroarch/cores folder in ArkOS and the ~/tmp/cores folder in 351Elec. Strangely, the easyrpg_libretro.info look nearly identical in terms of display_version = "0.6.1.0" However, easyrpg_libretro.so is 11.4MB on ArkOS and only 4.4MB in 351Elec. The file creation date is 11/11/20 in ArkOS and 4/22/21 in 351Elec.

The most recent references to any changes to the earyRPG core in the ArkOS changeling is March 8, 2021: https://raw.githubusercontent.com/christianhaitian/arkos/main/changelogs/rg351p-changelog. The source code for that update is available here: https://github.com/christianhaitian/arkos/releases/tag/v1.6.03.08.2021

Not sure if this information is helpful... but passing it along just in case. I don't know where that  6ad57329 hash is coming from either.

fry-guy, 2021-05-06:
P.S. Straight up swapping those core files from ArkOS for the ones in 351Elec did not solve the problem. So something else must be going on?

fry-guy, 2021-05-06:
No problem. Thanks again for looking into it!

fry-guy, 2021-05-06:
OK, I feel dumb. But try this?

In RetroArch, go to Settings --> Frame Throttle and select "Sync to Exact Frame Rate." (Or in your case, turn it on to create the problem, maybe?)

Could it be that this is just a preset issue? It's automatically off in ArkOS; it appears to be automatically on in 351Elec. 

Maybe it should be changed for this system only in future releases?

dhwz, 2021-05-06:
I can't remember that I've ever changed that setting.
I'll try it, thanks for the hint.

fry-guy, 2021-05-06:
It's under "Frame Throttle" in the main settings menu in RetroArch. Here's some photos: https://imgur.com/gallery/9zMV7Mk

I mean, I assume it was set "on" as the default. I don't remember even changing it either!

dhwz, 2021-05-06:
Sync to Exact Content Framerate you mean? That doesn't change anything for me? Does it change anything for you if you disable it?

fry-guy, 2021-05-06:
Oh, yes. Sync to Exact Content Framerate. I forgot the word "content."

Interesting... switching that literally solved the problem for me. And yet yours it makes no change? So strange.

What version of easyRPG are you running? Do I need to update something?

dhwz, 2021-05-06:
version0.6.2.3, you have to wait for the next update to use it

dhwz, 2021-05-06:
fixed with next release

closed 2021-05-06 by dhwz

fry-guy, 2021-05-06:
Awesome - thank you!

fry-guy, 2021-05-12:
Thank you so much for addressing this issue in your latest update!

Unfortunately, I am now encountering fatal errors when launching RPGMaker games after installing the latest version... Here is a photo of the error screen: https://imgur.com/gallery/u4DGDCK

I've also pasted my logfile below. This is occurring upon launching purchases copies of Ara Fell and Grimm's Hollow. I also tried a few other fanmade games, like the FF7 2D remake demo. Maybe there's a problem with my particular install? Any advice? (I already tried redownloading the RPG maker packages in the Downloads section.) 

Thank you again for all of your help/work on this...

[2021-03-20 22:35:13] Debug: EasyRPG Player 0.6.1 started
[2021-03-20 22:35:13] Debug: ============================
[2021-03-20 22:35:13] Debug: Seeded the RNG with 1616294111.
[2021-03-20 22:35:13] Debug: CLI: 
[2021-03-20 22:35:13] Debug: Framerate is 0 FPS!
[2021-03-20 22:35:14] Warning: Removing invalid party member 1
[2021-05-06 22:58:19] Debug: EasyRPG Player 0.6.1 started
[2021-05-06 22:58:19] Debug: ============================
[2021-05-06 22:58:19] Debug: Seeded the RNG with 1620356297.
[2021-05-06 22:58:19] Debug: CLI: 
[2021-05-06 22:58:19] Debug: Framerate is 0 FPS!
[2021-05-06 22:58:33] Warning: Removing invalid party member 1
[2021-05-12 17:55:24] Debug: EasyRPG Player 0.6.1 started
[2021-05-12 17:55:24] Debug: ============================
[2021-05-12 17:55:24] Debug: Seeded the RNG with 1620842121.
[2021-05-12 17:55:24] Debug: CLI: 
[2021-05-12 17:55:24] Debug: Framerate is 0 FPS!
[2021-05-12 17:55:53] Debug: Encoding not detected
[2021-05-12 17:55:53] Debug: Using /storage/roms/easyrpg/AraFell as Game and Save directory
[2021-05-12 17:55:53] Debug: Loading game Ara Fell
[2021-05-12 17:55:53] Debug: Game does not need RTP (FullPackageFlag=1)
[2021-05-12 17:55:53] Debug: Using RPG2k3 (English release, v1.11) Interpreter
[2021-05-12 17:55:53] Debug: RPG2k < v1.50 / RPG2k3 < v1.05 detected
[2021-05-12 17:55:53] Debug: Engine configured as: 2k=0 2k3=1 2k3Legacy=0 MajorUpdated=0 Eng=1
[2021-05-12 17:55:53] Debug: Adding /storage/roms/bios/rtp/2003/ to RTP path
[2021-05-12 17:55:54] Debug: RTP is "Official English" (675/675)
[2021-05-12 17:55:54] Debug: Cannot find: ./ExFont
[2021-05-12 17:55:54] Debug: Loading ExFont from /storage/roms/easyrpg/AraFell/RPG_RT.exe
[2021-05-12 17:55:54] Debug: EXEReader: BITMAP not found.
[2021-05-12 17:55:54] Debug: Actor 7: Removing item 136 from equipment slot 1 (Not equippable by this actor)
[2021-05-12 17:55:54] Debug: Actor 7: Removing item 381 from equipment slot 2 (Not equippable by this actor)
[2021-05-12 17:55:54] Debug: Actor 7: Removing item 346 from equipment slot 3 (Not equippable by this actor)
[2021-05-12 17:55:54] Debug: Actor 7: Removing item 422 from equipment slot 5 (Not equippable by this actor)

dhwz, 2021-05-12:
That's not the new version 0.6.2.3 you are launching? Are you sure you've updated to our new release?

fry-guy, 2021-05-12:
It should be? I put the tar file in the update folder and it said it was updating upon restart. I've also got the faster menus, etc.

Is there some reason the new core might not have installed despite the rest of it? Is there an updated easyrpg_libretro.so file somewhere I could just drop in place of an old one?

fry-guy, 2021-05-13:
OK, good news and bad news.

Good news: It works now! I flashed a new image of the latest OS on a fresh SD card, tested it, and it worked! Even better, I then grabbed easyrpg_libretro.so and easyrpg_libretro.info from tmp/cores on the new image and copied to tmp/cores on the old not working image, and I could launch the games there too! (I even confirmed that I am now running 0.6.3 under core information in RetroArch.) The battle images even work now in Grimm's Hollow! I dont know why, but it seems that the update did not replace the files in tmp/cores with the new files.

Bad news: There is a new bug in Ara Fell. The arrow keys don't work on the main menu and, when you do load a game, which you can do by hitting A before the intro screen finishes loading, the action key no longer works (shift or "Y" on the retropad). I played with a bunch of controller settings, and this simply seems unresolvable at the moment, which mean the game isn't playable since you can't navigate the maps without the ability to crouch with Y. The problem seems similar to this issue described here - https://github.com/EasyRPG/Player/issues/787 - which was supposedly resolved a few years ago. Should I create a new bug report?

dhwz, 2021-05-13:
Never ever play around with our provided cores, means don't touch that folder!

fry-guy, 2021-05-13:
Yep, will make sure not to going forward!

FYI tho that the control issues in Ara Fell exist on my card with the fresh install, too.

dhwz, 2021-05-13:
Yes as said before report bugs to the upstream repo.

fry-guy, 2021-05-13:
OK, I'll create a new report. Thanks!