AmberELEC/AmberELEC #355: Battle pictures not working in EasyRPG

fry-guy, 2021-05-07:
Me again. I've discovered another bug in EasyRPG. The core is not displaying battle pictures correctly, despite the fact this fix was supposedly implemented in the core in November 2008: https://github.com/EasyRPG/Player/issues/1488

(According to the EasyRPG developer's blog, "Pictures and Weather are now fully supported in battle scenes ([#1488](https://github.com/EasyRPG/Player/issues/1488), [#2003](https://github.com/EasyRPG/Player/pull/2003)). The picture support was the final step to fully supporting RPG Maker 2003 v1.12. ([#1263](https://github.com/EasyRPG/Player/issues/1263))" https://blog.easyrpg.org/

Here is a screenshot of the error messages in the game Grimm's Hollow: https://imgur.com/gallery/SAHkISM The  error messages  say "ShowPicture: Not supported in battle" and "ErasePicture: Not supported in battle."

Here is how the battle system show appear: https://cdn.cloudflare.steamstatic.com/steam/apps/1170880/ss_7566839189ffcc2e53c74e8b2a859e4c435d9db5.1920x1080.jpg?t=1604121984 The "battle image" of thee meter does not display correctly.

I am running firmware version 20210422, EasyRPG version 0.6.1  and EmulationStation 1.9.1.

Any help for this issue is greatly appreciated. Thank you.

dhwz, 2021-05-07:
please don't report new issues before you have the up-to-date version and tested it

closed 2021-05-07 by dhwz

fry-guy, 2021-05-07:
Ok, will do. I didn't realize the new version addressed other issues which weren't discussed.