AmberELEC/AmberELEC #378: Strange error after playing a particular RPG game (easyRPG)

brinaa, 2021-05-12:
Hello,

I just updated the latest version for 351ELEC. This issue I have is strange because I only experience this system error for this one particular game that I'm playing (Grimm's Hollow ver 1.1). I'm not exactly sure if this is because of the game itself or the system. For this game that I'm playing, the game runs fine but I notice some important game feature that weren't working correctly during battle so I thought maybe I should close the game and restart retroarch.
 
However, after I did that, I receive this system error "Software Failure. Check Log Files for more information. Guru Meditation #68B329DA9893E3409." The game works fine on my windows with no issue so it's only when I'm running the game on RG351M that I experience it. I also tried deleting the game and re-downloading it again, but I still experience the error in my console.

dhwz, 2021-05-12:
Just ignore the console error.
And if you have issues with games itself report that to the upstream repository.

fry-guy, 2021-05-12:
Hrm, I've encountered that same issue with the battle system in Grimm's Hollow. I think this is what you're describing: https://github.com/351ELEC/351ELEC/issues/355

I'm also getting  crashes on easyRPG after the update: https://github.com/351ELEC/351ELEC/issues/353#issuecomment-839991507

Following...

brinaa, 2021-05-13:
I think the next easy rpg update will fix this issue

closed 2021-05-13 by brinaa