AmberELEC/AmberELEC #384: Retrorun controls

sebvalmontx360, 2021-05-13:
### Describe the bug
I am using 351ELEC Reborn on RG351M. The new Retrorun emulator is very good for Dreamcast, Naomi, and Atomiswave. However, for Naomi and Atomiswave, the analog stick and the shoulder buttons do not work. The analog stick is essential for RG351M because its dpad is horrible. In addition, several games use 6 or 8 buttons. I could manage to enable the analog and shoulder buttons in ArkOS by editing the scripts that launch Retrorun for Naomi and AtomisWave. Can I do that for 351Elec reborn?

### To Reproduce
Steps to reproduce the behavior:
1. Go to Naomi or AtomisWave
2. Launch any game.

### Expected behavior
Analog and shoulder buttons do not work.

### Device (please complete the following information):
RG351M with 351ELEC reborn

navy1978, 2021-05-14:
I'm working on a similar case.
I have just delivered a fix for Atomiswave and Naomi games (for example Ikaruga) for tate mode (vertical games) enabling the right analog stick that was disabled and rotating the screen on the other side.
About the left analog stick, the situation is a littlebut different, certain games use it and someother not. In same gaems the DPAD is used for certain actions and the left analog stick is used for other actions (some 3D games). 
we can maybe add this option only on certain games checking the name of the rom that is execute, this means the rom should have the exect name of what we decide. An alternative would be to put those games for wich we want to enable the left analog stick in a specific folder (with a specific name)

sebvalmontx360, 2021-05-14:
I believe to adopt different folders would be an easier, but very efficient, solution.

sebvalmontx360, 2021-05-14:
If you decide to go through the name solutions, maybe you could adopt mame names, since these games can be found in the MAME 0.220 romset (and up).

navy1978, 2021-05-14:
In case of the second solution,  the folder should be a sub-folder something like this:
naomi/analog/

navy1978, 2021-05-14:
yes for the first solution I was thinking about that (MAME rimset names)

sebvalmontx360, 2021-05-14:
Both solutions would be effective.
The first (use file names) would be fancier, but seems to be harder to implement (at least to me).
The second one (folders) would work anyway.

Thank you for the fast response.

navy1978, 2021-05-14:
I can work on this during week-end and see what I can do

sebvalmontx360, 2021-05-14:
I was thinking and, maybe, the folder would be more interesting to the user. 

As I said, RG351M DPad is horrible, which means that I want to use analog even in games it is not necessary (fighting games, for instance). 

If I have the folders, I could choose when to map dpad -> analog and when to use analog as analog.

sebvalmontx360, 2021-05-14:
Once again, thank you very much.

sebvalmontx360, 2021-05-14:
Would it be possible to add a parameter to Retrorun in ES frontend (map dpad to analog or something like it)? If yes, it would maintain the standard of the 351ELEC amazing interface, which allows to set several  emulator parameters very easily in the interface.

navy1978, 2021-05-14:
that is already there, I'm talking about a solution that works out of the box

sebvalmontx360, 2021-05-14:
Sorry but I could not find it. Once I open game setting and choose Retorun32, I can only choose the core. I do not see any additional option.

navy1978, 2021-05-14:
Ah I missunderstood you , I'm not talking about the 351elec interface but about the library itself.

navy1978, 2021-05-14:
Maybe both solution can be implmemented, and the user can choose

sebvalmontx360, 2021-05-14:
It would be amazing if you could do that.

For now, should updates be installed manually, correct?

navy1978, 2021-05-14:
it's planned to integrate retrorun in the distro so in the future no package would be needed.

sebvalmontx360, 2021-05-14:
Thank you. The improvements of 351ELEC Reborn are really amazing.

An improvement suggestion would be to include a faster standalone emulator for N64. Retroarch cores does not run very well in RK3326 chipset.

navy1978, 2021-05-14:
Trying to use retrorun with n64 and Saturn emulator is in my todo-list as well

sebvalmontx360, 2021-05-14:
Congratulations from your amazing job. 351ELEC Reborn made me moving from ArkOS.

damiansandoval, 2021-09-01:
This seems to be happen also with RGB351V using retrorun for Atomiswave games. For example I launch Dolphin Blue and the analog sticks is dead, oply DPAD works. Is there any configuration or opcion to modify?

I'm on RG351V with 351ELEC Firmware

closed 2021-11-10 by dhwz