AmberELEC/AmberELEC #385: es_systems_custom.cfg not recognized

Manfred282, 2021-05-13:
Describe the bug

old "es_systems.cfg" is renamed to "es_systems.oldcfg-rename-to:es_systems_custom.cfg-if-needed" after the last update.

if i rename this file to "es_systems_custom.cfg" it isnt recognized by emulationstation.

i use the "es_systems.cfg" to manually sort the systems, because in emulationstation there is no option to manually
sort systems. rename to "es_systems_custom.cfg" does nothing, rename to "es_systems.cfg" works till the next reboot
of the device, after that its "es_systems.oldcfg-rename-to:es_systems_custom.cfg-if-needed" again

To Reproduce

Steps to reproduce the behavior:

rename "es_systems.oldcfg-rename-to:es_systems_custom.cfg-if-needed" to "es_systems_custom.cfg"

Device (please complete the following information):

RG351P - 351ELEC - 20210510

konsumschaf, 2021-05-13:
Could you please rename it to `es_systems_test.cfg`?

Manfred282, 2021-05-13:
i did, doesnt help at all, sort order wrong, so its not recognized

konsumschaf, 2021-05-13:
OK, next step.´: Could you check for errors ins this file? The additional files are completely ignored, if they contain any error. 

You can check the `es_log.txt` file in `/tmp/logs` like this (I created a faulty file to reproduce this):

```
~ # grep parse /tmp/logs/es_log.txt
2021-05-13 18:31:51     ERROR   Could not parse es_systems_faulty.cfg file!
```

Manfred282, 2021-05-13:
no errors about the es_systems so far, take a look here. i removed just the no games warning log entries.

[image]

konsumschaf, 2021-05-13:
Please rename the file again to `es_systems_custom.cfg`, restart ES and then enter `grep parse /tmp/logs/es_log.txt` on the ssh shell on the system. ES is not as picky with `es_systems.cfg` as with any additional file.

konsumschaf, 2021-05-13:
You seem to have the same problem as a user on discord. 

> OK, I think this is due to the way the system parses the configs. If there is no user es_systems.cfg it will use the system wide one - and only changes the part of the xml tree it finds in additional custom files - not altering the order anymore. We can change the renaming part to only happen after an update. Thus you still have to rename the file after every update, but not after every boot. If we skip this part no user will see new cores etc. after an update ...

I will change the behaviour of the system, to only rename the file after an update - I hope this works for you.

Manfred282, 2021-05-13:
okay, will wait for a update, till then i have to live with a unsorted list ;) thanks

closed 2021-05-13 by konsumschaf