AmberELEC/AmberELEC #386: Controller issue in EasyRPG

fry-guy, 2021-05-13:
There appears to be a controller bug in EasyRPG core in the new update... In Ara Fell (full version, Steam), the arrow keys no longer work on the main menu and also the Y key ("shift") no longer works. This means the game isn't playable since you can't navigate the maps without the ability to crouch. You also can't even load your save games unless you hit "A" immediately after the developer splash screen, which jumps you straight to the "continue" option, as otherwise that option is unselectable due to the arrow key issue.

I played with a bunch of controller settings, and this simply seems unresolvable at the moment, although there may be something I am overlooking. This bug did not exist in the previous version of EasyRPG (although obviously the new version resolves some more serious performance issues!)

One thought I had: Could a keyboard control option be added to the core? Right now, you can only select RetroPad or RetroPad with Analog. I wonder if the ability to map keys to a keyboard input might override the issue?

I am using the latest version of the firmware: 20210510. EasyRPG core version 0.6.2.3. Emulation Station version 1.9.2.

The problem seems very similar to this issue described here - EasyRPG/Player#787 - which was supposedly resolved a few years ago. This post also suggests there may be some issue in RPGMaker called the "stuck key glitch"? https://steamcdn-a.akamaihd.net/steam/apps/440540/manuals/AraFellREADME.pdf?t=1461965438

Thanks for looking into this, as well as your continued amazing work on 351Elec!

Ghabry, 2021-05-14:
There is a core option where you can set if you want keyboard passtrough or not.

fry-guy, 2021-05-14:
How do you set up keyboard option? I only see RetroPie and retro pad + analog under "controls" in RetroArch settings.

dhwz, 2021-05-14:
Didn't I say you have to report bugs to upstream? We can't fix issues inside the emulators.

Ghabry, 2021-05-14:
Imo this is not a bug but a user error ;)

I said "Core Options". These core specific settings you get when you load a game in there, pause and select "Options" (or how this was called). THere is an option to set  the input to "Keyboard" or "RetroPad" or "Both".

dhwz, 2021-05-14:
I don't know, I'm just confused as he said it worked before our update.

fry-guy, 2021-05-14:
I think I have both of those options activated but I will double check. I know switching those on and off was one of the first things I tried when I first started having problems, so I'm pretty confident it didn't make a difference... 

I may be confused by the term "upstream" here. I thought you meant don't keep posting to the other report since it was closed. 

I can also post this over on the easyrpg GitHub page. It is accurate that I didn't encounter this particular error until after the update.

Ghabry, 2021-05-14:
Reported upstream please close this one here

closed 2021-05-14 by anthonycaccese