AmberELEC/AmberELEC #390: sdlpop is too slow.

Praeambulum, 2021-05-14:
351elec is good but sdlpop is too slow. Also, so many frame drops.
I don't know why but in emuelec sdlpop is very smooth and fast.

dhwz, 2021-05-16:
Yes seems to be not working good at all as the whole gui in menu is also not working properly.
I'll build the current version and test if that helps.

dhwz, 2021-05-16:
Sorry doesn't seem to be fixable right now, I'll disable our port. Please try to use the Anberports version afterwards.

closed 2021-05-16 by dhwz

Tokodomo, 2021-05-31:
The reason is a faulty or improper build/version of SDL (e.g. libSDL2-2.0.so). If you use libSDL2-2.0.so from: https://github.com/krishenriksen/AnberPorts/raw/master/misc/sdlpop.zip package, all (OpenTyrian also) is running at normal speed again. So for testing purposes you can extract libSDL2-2.0.so from sdlpop.zip and add:
`export LD_LIBRARY_PATH=/your/custom/path/to/libSDL2-2.0…`
to "Prince of Persia.sh",  "OpenTyrian.sh" etc. to bring back the normal/full game speed.