AmberELEC/AmberELEC #392: Error deleting themes in RG351M

sebvalmontx360, 2021-05-14:
### Describe the bug
I cannot remove themes that I have installed in 351ELEC Reborn for RG351M. 

### To Reproduce
Steps to reproduce the behavior:
1. Press Start
2. Go to Dowloads -> Themes
3. Select any downloaded theme
4. Try to remove the theme

### Expected behavior
At first, I flashed 351ELEC Reborn from scratch. Then I installed all themes currently available in order to check each one. After checking them I tried to delete some themes but I received an error message saying that the theme could not be removed. I received the same error message for 3 or 4 different themes. In addition, I was not trying to delete the theme I was currently using.

### Device (please complete the following information):
I am using 351ELEC Reborn in RG351M.

Dahak66, 2021-05-14:
Just adding that this is also the case on the 351V as well, and has been this way at least since the previous (formerly final) build. Deletion from within the file manager does work, though, and the download script does recognise deleted themes as available to download.

sebvalmontx360, 2021-05-14:
Thank You. Could You tell me where the themes are stored?

Dahak66, 2021-05-15:
From the filemanager, go to /etc/emulationstation/themes.

closed 2021-07-21 by dhwz