AmberELEC/AmberELEC #396: Easy RPG graphical issues and game malfunctions

MikhailovDenegir, 2021-05-15:
Hello,

Thanks for your continued work on 351ELEC. I wanted to bring an issue with EasyRPG on RetroArch. With the proper rtp files installed, most games have graphical issues and don’t function properly. For example, in Grimm’s Hollow, some graphics don’t display properly and in battle, you don’t earn spirit points for hitting an enemy. I am unsure if this issue is config related or related to the core but I thought I would let you know. The emulator is very much improved from the last release.

dhwz, 2021-05-16:
Please report problems related to the emulator itself to the easyrpg repo, we can't help or fix anything related to this.

closed 2021-05-16 by dhwz

MikhailovDenegir, 2021-05-16:
Did as you asked. The issue is linked below. 

[](https://github.com/libretro/easyrpg-libretro/issues/53)

Will most likely be resolved in a future update of the core in July. Just posting here in case you need to include the eventual updated core in future builds of 351ELEC.