AmberELEC/AmberELEC #406: Core requests

nl255, 2021-05-17:
I would like to see the following cores added if it is practical to do so.  Especially SMS Plus GX.

SMS Plus GX
Mame 2000 (mame4all)
Quicknes
Mednafen supafaust
PCSX Rearmed Neon

dhwz, 2021-05-18:
I'm currently reviewing the available cores, I'll add this one to my list.

dhwz, 2021-05-19:
SMS Plus GX (yes)
Mame 2000 (mame4all) (for what is this one needed?)
Quicknes (doesn't work)
Mednafen supafaust (yes)
PCSX Rearmed Neon (never heard of that one?)

nl255, 2021-05-21:
> 
> 
> SMS Plus GX (yes)
> Mame 2000 (mame4all) (for what is this one needed?)
> Quicknes (doesn't work)
> Mednafen supafaust (yes)
> PCSX Rearmed Neon (never heard of that one?)

For Mame 2000/mame4all many lower end devices use that version and it would be nice to be able to use the same romset and high score data on the 351.  As for PCSX Rearmed Neon I found the .info file in the 351elec image which I have attached as a text file.

[pcsx_rearmed_libretro_neon.txt](https://github.com/351ELEC/351ELEC/files/6525416/pcsx_rearmed_libretro_neon.txt)

dhwz, 2021-05-22:
That's what we already have neon is the gpu of our arm processor.

closed 2021-05-25 by dhwz