AmberELEC/AmberELEC #411: PC Engine CD/ TG16 CD games does not work

YuuichirouR, 2021-05-18:
PC Engine CD games does not run. Got Guru meditation error.

###To Reproduce
Steps to reproduce the behavior:
1. Put .chd or cue/iso image in folder roms\pcenginecd or roms\tg16cd (I tried Lords Of Thunder, Castlevania, Gradius 2)
2. Put syscard3.pce (tried Japanese and US) in bios folder
3. Try to run any game 
4. Error "System Failure. Check Log Files for more information. Guru Meditation #68B329DA9893E3409" appears and return to game menu

### Expected behavior
Game runs well

### Device (please complete the following information):
 - 351ELEC - 20210510 Reborn Release on RG351p, flashed from scratch.

bebeidon, 2021-05-18:
I don't think this is a bug, it works as expected for me, tested various roms and both cores.

YuuichirouR, 2021-05-18:
Here logs after I tried to run another game.
```
2021-05-19 00:55:24	INFO	SDL AUDIO Initialized
2021-05-19 00:59:40	INFO	 req sound [basic.launch]
2021-05-19 00:59:40	INFO	   (missing)
2021-05-19 00:59:40	INFO	Attempting to launch game...
2021-05-19 00:59:40	INFO	Attempting to close SDL AUDIO
2021-05-19 00:59:41	INFO	SDL AUDIO Deinitialized
2021-05-19 00:59:41	INFO	Configure emulators command : -p1index 0 -p1guid 03000000091200000031000011010000 
2021-05-19 00:59:41	INFO		/usr/bin/runemu.sh /storage/roms/pcenginecd/Ai_Chou_Aniki_\(NTSC-J\)_\[NSCD5018\].chd -Ppcenginecd --core=mednafen_pce_fast --emulator=libretro --controllers="-p1index 0 -p1guid 03000000091200000031000011010000 "
2021-05-19 00:59:47	WARNING	...launch terminated with nonzero exit code 256!
2021-05-19 00:59:47	INFO	Window::init
2021-05-19 00:59:47	INFO	Creating window...
2021-05-19 00:59:47	INFO	Created window successfully.
2021-05-19 00:59:47	INFO	Checking available OpenGL extensions...
2021-05-19 00:59:47	INFO	 ARB_texture_non_power_of_two: MISSING
2021-05-19 00:59:47	ERROR	Could not find joystick to close (instance ID: 1)
2021-05-19 00:59:47	INFO	Added known joystick OpenSimHardware OSH PB Controller (instance ID: 1, device index: 0)
2021-05-19 00:59:48	ERROR	CECInput::mAdapter->DetectAdapters failed
2021-05-19 00:59:48	INFO	SDL AUDIO Initialized
2021-05-19 00:59:48	WARNING	...Show Error message! exit code 256!
2021-05-19 00:59:48	INFO	Attempting to close SDL AUDIO
2021-05-19 00:59:49	INFO	SDL AUDIO Deinitialized
2021-05-19 01:00:02	INFO	Window::init
2021-05-19 01:00:02	ERROR	Could not find joystick to close (instance ID: 2)
2021-05-19 01:00:02	INFO	Added known joystick OpenSimHardware OSH PB Controller (instance ID: 2, device index: 0)
2021-05-19 01:00:02	ERROR	CECInput::mAdapter->DetectAdapters failed
2021-05-19 01:00:03	INFO	SDL AUDIO Initialized
```

bebeidon, 2021-05-18:
did you check if the rom is ok by running it on your pc?

Cebion, 2021-05-18:
Please check the retroarchlogs for error messages why it didn't load.
Visit us in our discord for support =)

closed 2021-05-18 by Cebion