AmberELEC/AmberELEC #416: Guru meditation error in several emulators

Blesaster, 2021-05-21:
## This is only for actual bugs that occur on the system! For Support visit us in our Discord
https://discord.gg/bmXtCt88Tz

### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.

### Describe the bug
whenever I try to play any Atari2600 game I get the message:

"System Failure. Check Log Files for more information. Guru Meditation #68B329DA9893E3409" 

I tried to change the core but the error persisted. Then the same error started happening with other emulators.

### To Reproduce
Steps to reproduce the behavior:
1. Go to Atari 2600
2. Click on any game
3. Then I get a blank screen for a while and the Guru meditation error is written on the screen. The device is frozen for a while and the goes back to 351Elec
4. See error

"System Failure. Check Log Files for more information. Guru Meditation #68B329DA9893E3409" 

### Expected behavior
It should start the game

### Device (please complete the following information):
 - 351ELEC Reborn 20210510 on an RG351M with a 128Gb Sandisk extreme micro SD card

### Additional context
Since the appearance of this error some other games stopped working, giving me a black screen and going back to Emustation, usualy with games I've played before, with new games it doesn't happen.

konsumschaf, 2021-05-21:
Try to re-add one of a non working game under a different file-name - feels like a dying sd card to me - feel free to join our discord - it's easier to discuss these issues there.

Blesaster, 2021-05-21:
Thanks for the reply… the card is brand new… maybe I should create it again.

Regarding Discord… I’m a noob there… I obly see the anouncements but nowhere to post and ask questions..

Cebion, 2021-05-21:
You need to read the rules and react to the message in the rules.

closed 2021-05-21 by Cebion