AmberELEC/AmberELEC #419: RG351V: Missing on-screen keyboard for game meta data

mtoensing, 2021-05-23:
### Describe the bug
When you edit a games meta data there is no option to edit the text. This was possible in older versions of emulec. 

### To Reproduce
Steps to reproduce the behavior:
1. Go to game list
2. Press X
3. Edit game meta data
4. Unable to edit this text because no on screen keyboard is visible. I tried all buttons.

### Expected behavior
on screen keyboard allows to edit the text

### Screenshots
[image]

### Device (please complete the following information):
 - 351ELEC 20210510 Reborn Release

Cebion, 2021-05-23:
I wonder if this has ever worked on 351elec

mtoensing, 2021-05-29:
Keyboard is missing in all menus (wifi, passwords…) had to reflash the Image. This happened after the update to reborn.

dhwz, 2021-07-16:
invalid issue

closed 2021-07-16 by dhwz

mderie, 2022-12-24:
Hi,
Why it is invalid ? I had the same issue with the RG351M when entering the wifi password.
I had to edit the file under Linux on my laptop...
Regards