AmberELEC/AmberELEC #420: ECWolf libretro core

masid101, 2021-05-23:
Hello,

any chance of adding this core (maybe ports). I would love to play the good old Wolfenstein 3D and Spear of Destiny.

Marxiboy, 2021-05-23:
They are already in Anberports. See https://github.com/krishenriksen/AnberPorts

dhwz, 2021-05-23:
sure why not let me check if the core builds

masid101, 2021-05-23:
I just tried the AnberPorts version and it doesn't work for me. I think the libretro version would be far more elegant and usable.

tiduscrying, 2021-05-23:
I didn't do a whole lot as far as actual port work is concerned, but I did contribute it to Anberports. I don't have a 351V to test it with but I don't see any reason it wouldn't work. Have you tried using the vanilla, non Anberports release from the normal repo?

https://github.com/tiduscrying/ecwolf-rg351

dhwz, 2021-05-23:
We don't need that anymore ecwolf is now a integrated system.

masid101, 2021-05-24:
Thank you very much, can't wait to try out the next release and test this.

closed 2021-05-24 by masid101