AmberELEC/AmberELEC #423: Savestates saving two folders deep

cscryan, 2021-05-23:
### Describe the bug
Savestates seem to create an unneeded folder with the same system name within the appropriate directory.

(i.e., savestates are saving one folder deeper than they should -- /storage/roms/savestates/nes/nes/Game.state.auto)

### To Reproduce
Steps to reproduce the behavior:
1. Start a game
2. Save using the save state shortcut (select + R2 for me)

### Expected behavior
Save states are organized into the appropriate folders.

### Screenshots
https://imgur.com/a/0PFn2Xq

### Device (please complete the following information):
 - RG351v

baskapteijn, 2021-05-28:
This does not happen for me with `20210510 Reborn Release` on `RG351V` for any of the consoles, including``NES`. @cscryan which release are you using?

Cebion, 2021-06-10:
Sorry, this is no support forum.

For Support please visit us on our Discord Server
https://discord.gg/bmXtCt88Tz

Only create issues if it is reproducable issue.

closed 2021-06-10 by Cebion