AmberELEC/AmberELEC #424: Psx wont show

tufttt, 2021-05-23:
Psx isn't showing on the main system menu on my rg351v and i do have it selected in the games collection settings to be displayed. I can play psx games through retroarch when i open them manually but its very inconvenient.

dhwz, 2021-05-23:
You are missing your cue files for your bin images?
Unhide the bin extension or better create the missing cue files.

closed 2021-05-23 by dhwz

tufttt, 2021-05-24:
Do i put the bin. File and the cue. File in a folder then put them in the psx folder or do i just put them in the psx folder separately?

konsumschaf, 2021-05-24:
Just put everything together in the psx folder. And have a look at .chd. ;-)

tufttt, 2021-05-27:
What do i do with games like resident evil 2 that has 2. Bin files?

sertiana, 2021-05-27:
You can create a .m3u playlist

tufttt, 2021-05-27:
Sorry for asking so many questions but how do i do that?

Cebion, 2021-05-27:
Please visit our discord for support.
This is no support forum.