AmberELEC/AmberELEC #425: too many "In game" Brightness steps

dskmusic, 2021-05-24:
## This is only for actual bugs that occur on the system! For Support visit us in our Discord
https://discord.gg/bmXtCt88Tz

### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.

### Describe the bug
when you are in a game, presing L3+ L1 or L2 to decrease/increase brigthness have a lot of steps!

### To Reproduce
Steps to reproduce the behavior:
1. Go to any game
2. Use combination keys to modify brightness
3. You can see too many steps... too many!!

### Expected behavior
Few steps to modify the in game brightness

### Device (please complete the following information):
 - RG351M last release, "Reborn"

dhwz, 2021-05-24:
This is not a bug

closed 2021-05-24 by dhwz

pumpkinlink, 2022-03-04:
@dskmusic I've implemented a change for a 10 of 100 step instead of 1 of 100 on my fork. Did not open a PR because it currently breaks current functionality on devices with volume buttons like RG351V and MP. You can use my script without having to compile a firmware image, just edit the `storage/.config/custom_start.sh` as explained on this gist (read the comments below): https://gist.github.com/pumpkinlink/ca21994edbb76710af799b7c54bfb4fc