AmberELEC/AmberELEC #427: cannot access the mainmenu for psp game like : Avatar James Cameron The Game

NguyenThang45, 2021-05-25:
## This is only for actual bugs that occur on the system! For Support visit us in our Discord
https://discord.gg/bmXtCt88Tz

### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.

### Describe the bug
A clear and concise description of what the bug is.

### To Reproduce
Steps to reproduce the behavior:
1. Go to the game ( Avatar James Cameron The Game for PSP )
2. Click on Start until it done loading and acces to main menu to create profile to start game
3. see the error that the main menu no display, only blank screen.

### Expected behavior
i want to know  the main root of problem and how to solve it.

### Screenshots
If applicable, add screenshots to help explain your problem.

### Device (please complete the following information):
 - RGB 351P
### Additional context
cannot play the game because the main menu is a blank. Was try different  emulator but not working.
pls contact me thourgh email.

dhwz, 2021-05-25:
This is a emulator related issue please report that to the repository of the used emulator, we can't help.

closed 2021-05-25 by dhwz

NguyenThang45, 2021-05-25:
can you specify who can help me ? , please