AmberELEC/AmberELEC #430: eduke32 port broken in RG351V

mtoensing, 2021-05-25:
### Describe the bug
eduke32 (Duke Nukem 3D) won't start with either the Atomic Edition GRP/RTS files nor the Trial GRP/RTS files. 

### To Reproduce
Steps to reproduce the behavior:
1. put DUKE.RTS and DUKE3D.GRP in /storage/roms/ports 
2. Go to "ports"
3. Scroll down to '.Duke Nukem 3D..'
4. black screen

### Expected behavior
Duke Nukem 3D should run with the above-mentioned files.

### Log
Log seems fine: 
```
EDuke32 r9309-d2ee87aea
Built May 11 2021 01:58:21, GCC 10.2.1, 64-bit 
Using /storage/roms/ports/eduke/ for game data
Using /storage/.config/eduke32/ for game data
Initializing SDL 2.0.14
Using "KMSDRM" video driver
Searching for game data...
Using "duke3d.grp" as main game data file.
Compiling: GAME.CON (151190 bytes)
Including: DEFS.CON (35992 bytes)
Including: USER.CON (45482 bytes)
USER.CON:844: warning: sound 261 already defined (hydro43.voc)
USER.CON:873: warning: sound 323 already defined (aisle402.voc)
GAME.CON: In actor `ORGANTIC':
GAME.CON:3967: warning: found `else' with no `if'
GAME.CON: In state `pigshootenemystate':
GAME.CON:5890: warning: found `else' with no `if'
Found 4 warning(s), 0 error(s).
Compiled 140344 bytes in 51ms
Initialized 96.0M cache
Sound hlminhal.voc(#23) not found!
Sound hlmexhal.voc(#24) not found!
Sound yes.voc(#45) not found!
Sound fan.voc(#77) not found!
Sound hover.voc(#80) not found!
Sound turrrm.voc(#145) not found!
Sound turrrg.voc(#146) not found!
Sound turrat.voc(#147) not found!
Sound b3atk01.voc(#152) not found!
Sound turrpn.voc(#161) not found!
Sound turrdy.voc(#162) not found!
Sound b3atk01.voc(#176) not found!
Sound vpiss2.voc(#182) not found!
Sound jacuzzi2.voc(#322) not found!
Sound joke.voc(#397) not found!
RTS file "DUKE.RTS" loaded
Initializing OSD...
Game controllers:
  1. OpenSimHardware OSH PB Controller
Using controller OpenSimHardware OSH PB Controller
Executing "settings.cfg"
Setting video mode 640x480 (8-bpp fullscreen)
Refresh rate: 149.00Hz
videoSetGamma(): That operation is not supported
videoSetGamma(): That operation is not supported
videoSetGamma(): That operation is not supported
Initializing sound: SDL ALSA driver on rockchip,rk817-codec, : 48.0 KHz stereo with 64 voices
Initializing MIDI driver: AdLib OPL3 emulation
 
Wrote eduke32.cfg
Wrote settings.cfg
```

the files in /storage/.config are generated as well: 

settings.cfg
grpfiles.cache
eduke32.cfg

### Device (please complete the following information):
351ELEC - 20210510 Reborn Release on RG351V

Tokodomo, 2021-05-31:
Eduke32 is very greedy and needs X amount of memory to run. So we need to increase swap size to 300M (or better 384M to be on the safe side), at least on devices with 1GB of RAM.

Before:
```
~ # swapon --show
NAME                     TYPE SIZE USED  PRIO
/storage/.cache/swapfile file 128M   0B 10000
```
Changing swap size:
```
~ # swapoff /storage/.cache/swapfile
~ # dd if=/dev/zero of=/storage/.cache/swapfile bs=1024 count=384k
~ # mkswap /storage/.cache/swapfile
~ # sync
~ # swapon -p 10000 /storage/.cache/swapfile
```
After:
```
~ # swapon --show
NAME                     TYPE SIZE USED  PRIO
/storage/.cache/swapfile file 384M   0B 10000
```

mtoensing, 2021-06-01:
I tried these steps but it still does not start.

Tokodomo, 2021-06-01:
Okay. For me it works after these steps. I got a segfault before any other output:
```
~ # cd /roms/ports/eduke
/roms/ports/eduke # eduke32 
Segmentation fault
```
Now it runs flawlessly (after increasing swap size to 384M).

mtoensing, 2021-06-01:
Did you start it via ssh or via the menu?

Tokodomo, 2021-06-01:
I tried both (menu and ssh), but no output at all (even `/tmp/logs/exec.log` was empty). I hadn’t noticed any log in the game dir (maybe there was none at the time though but I’m not sure). I use the .grp and .rts files from the 1996 release.

mtoensing, 2021-06-02:
I even tried 500 MB
```
swapon --show
NAME                     TYPE SIZE USED  PRIO
/storage/.cache/swapfile file 500M   0B 10000
```
Does not work. But this is the normal DUKE3D.GRP file of the lastest version of Duke Nukem (1.5, Atomic) from 1997.

```
~/roms/ports/eduke # eduke32
EDuke32 r9309-d2ee87aea
Built May 11 2021 01:58:21, GCC 10.2.1, 64-bit 
Using /storage/roms/ports/eduke/ for game data
Using /storage/.config/eduke32/ for game data
Initializing SDL 2.0.14
Using "KMSDRM" video driver
Searching for game data...
Using "DUKE3D.GRP" as main game data file.
Compiling: GAME.CON (151190 bytes)
Including: DEFS.CON (35992 bytes)
Including: USER.CON (45482 bytes)
USER.CON:844: warning: sound 261 already defined (hydro43.voc)
USER.CON:873: warning: sound 323 already defined (aisle402.voc)
GAME.CON: In actor `ORGANTIC':
GAME.CON:3967: warning: found `else' with no `if'
GAME.CON: In state `pigshootenemystate':
GAME.CON:5890: warning: found `else' with no `if'
Found 4 warning(s), 0 error(s).
Compiled 140344 bytes in 62ms
Initialized 134.1M cache
Sound hlminhal.voc(#23) not found!
Sound hlmexhal.voc(#24) not found!
Sound yes.voc(#45) not found!
Sound fan.voc(#77) not found!
Sound hover.voc(#80) not found!
Sound turrrm.voc(#145) not found!
Sound turrrg.voc(#146) not found!
Sound turrat.voc(#147) not found!
Sound b3atk01.voc(#152) not found!
Sound turrpn.voc(#161) not found!
Sound turrdy.voc(#162) not found!
Sound b3atk01.voc(#176) not found!
Sound vpiss2.voc(#182) not found!
Sound jacuzzi2.voc(#322) not found!
Sound joke.voc(#397) not found!
Initializing OSD...
Game controllers:
  1. OpenSimHardware OSH PB Controller
Using controller OpenSimHardware OSH PB Controller
Executing "settings.cfg"
Setting video mode 640x480 (8-bpp fullscreen)
Refresh rate: 149.00Hz
videoSetGamma(): That operation is not supported
videoSetGamma(): That operation is not supported
videoSetGamma(): That operation is not supported
Initializing sound: SDL ALSA driver on rockchip,rk817-codec, : 48.0 KHz stereo with 64 voices
Initializing MIDI driver: AdLib OPL3 emulation
eduke32[0x56795c]
linux-vdso.so.1(__kernel_rt_sigreturn+0x0)[0x7f81070680]
/usr/lib/libc.so.6(strnlen+0x20)[0x7f809d1ec0]
/usr/lib/libc.so.6(strncpy+0x24)[0x7f809d2274]
eduke32[0x478e60]
eduke32[0x4378cc]
/usr/lib/libc.so.6(__libc_start_main+0xe0)[0x7f80978270]
eduke32[0x406afc]
 
Wrote eduke32.cfg
Wrote settings.cfg
```

DominikPiskorz, 2021-06-13:
I had the same issue with eduke32, with exact same log. After updating to newest 351Elec it now launches properly

mtoensing, 2021-06-15:
Yes, same here. Controls are broken on RG351V but it starts.

closed 2021-06-15 by mtoensing