AmberELEC/AmberELEC #431: Makefile Error Building 351ELEC (Arch Linux System)

rstrube, 2021-05-26:
Hello Everyone,

I realize the recommended build environment is Ubuntu, but I was hoping to get everything setup so I can build on my Arch Linux box.  I believe I have all the necessary prerequisites installed (I mapped everything over from the Ubuntu packages).  I'd also be happy to submit a PR with build instructions for Arch (when I hopefully get everything working!).

For reference here are the packages that I installed (some were already installed on my system):

```
pacman -S gcc make git unzip wget xz sdl2 sdl2_mixer freeimage freetype2 curl libcurl-gnutls rapidjson alsa-lib mesa base-devel boost-libs cmake ttf-droid vlc texinfo premake go patchelf xmlstarlet patchutils gawk gperf libxfont2 jre-openjdk python libxslt perl-json lzop ncurses dtc uboot-tools rsync p7zip unrar perl-parse-yapp zip aarch64-linux-gnu-binutils dos2unix openssl perl-xml-parser bc xorg-bdftopcf xorg-mkfontscale libcurl-compat
```

Unfortunately, I'm running into a Makefile error (within the `make-4.3` directory):

```
make[1]: *** [Makefile:1086: /home/robert/Projects/351ELEC/build.351ELEC-RG351P.arm-20210510/make-4.3/Makefile.in] Error 1
```

I dug in a bit, and the relevant line in that make file is:

```
@for dep in $?; do \
```

For larger context here is some of the surrounding lines:

```
MAKETESTFLAGS =
all: all-recursive
 
.SUFFIXES:
.SUFFIXES: .c .o .obj
am--refresh: Makefile
        @:
$(srcdir)/Makefile.in:  $(srcdir)/Makefile.am  $(am__configure_deps)
        @for dep in $?; do \
          case '$(am__configure_deps)' in \
            *$$dep*) \
              echo ' cd $(srcdir) && $(AUTOMAKE) --foreign'; \
              $(am__cd) $(srcdir) && $(AUTOMAKE) --foreign \
                && exit 0; \
              exit 1;; \
          esac; \
        done; \
        echo ' cd $(top_srcdir) && $(AUTOMAKE) --foreign Makefile'; \
        $(am__cd) $(top_srcdir) && \
          $(AUTOMAKE) --foreign Makefile
Makefile: $(srcdir)/Makefile.in $(top_builddir)/config.status
        @case '$?' in \
          *config.status*) \
            echo ' $(SHELL) ./config.status'; \
            $(SHELL) ./config.status;; \
          *) \
            echo ' cd $(top_builddir) && $(SHELL) ./config.status $@ $(am__maybe_remake_depfiles)'; \
            cd $(top_builddir) && $(SHELL) ./config.status $@ $(am__maybe_remake_depfiles);; \
        esac;
```

It almost appears that something about that syntax is causing build process to fail.  Perhaps the versions of my GNU autotools are incompatible with some of the syntax in this particular makefile? Any advice/suggestions from the developers would be greaty appreciated!

Thanks!

dhwz, 2021-05-26:
Use docker, see #429

closed 2021-05-26 by dhwz

rstrube, 2021-05-26:
@dhwz Thanks, I saw this PR shortly after I opened my issue, and it's a great way to standardize the build environment.  I was able to setup docker and I'm currently building 351ELEC!  Really excited to check out all the recent changes.