AmberELEC/AmberELEC #436: Connecting to WiFi Network the First Time Causes UI to Freeze

rstrube, 2021-05-26:
### Describe the bug
Connecting to WiFi network for the first time causes the UI to freeze.  You must toggle the power button on the RG351P to restore UI interaction.

### To Reproduce
Steps to reproduce the behavior:
1. Go to Network Settings
2. Setup your WiFi SSID and WiFi Key
3. After entering the WiFi Key the system will attempt to connect to the WiFi network.
4. The UI will freeze
5. Toggling the power button will cause the UI to un-freeze and you'll be presented with a dialog box telling you the WiFi has been connected.  UI interaction is restored.

### Expected behavior
The UI shouldn't freeze after entering WiFi credentials.  You shouldn't have to toggle the power button for the UI to become responsive again.

### Device (please complete the following information):
 - RG351P device
 - Issue present on 20210510 Reborn Release
 - Issue present building from `main` as of commit af0da95ba61896c20b969267cdec3e6671bd57d7

baskapteijn, 2021-05-27:
I want to add that this also happens with:

- RG351V
- On 20210510 Reborn Release

szalik-rg351, 2021-05-27:
truth with wifi freeze i have same thing

baskapteijn, 2021-06-05:
This also happens with a fresh install of release `351ELEC-RG351V.aarch64-20210603.img.gz`, This may be stating the obvious but I think it's relevant info nevertheless.

rstrube, 2021-07-14:
I just conducted a fresh install of `351ELEC-RG351M.aarch64-20210603` (hotfix 3) and the UI freeze felt more reasonable. The dialog box that pops up telling the user that you've successfully connected appeared after 4-5 seconds.  During that time the UI was "frozen" but I didn't have to toggle the power button like I had to in the past.  Perhaps this is something that should be addressed upstream with Emulation Station? (e.g. adding some sort of visual feedback that an action is occurring in the background).

dhwz, 2021-09-28:
can't reproduce

closed 2021-09-28 by dhwz

janbiasi, 2021-12-06:
Experienced the same issue with the RG351MP, running version Pineapple (20211122). The workaround which was described by  @rstrube works indeed but isn't a solution for first-time connection imo.