AmberELEC/AmberELEC #439: At low power, system won’t boot properly

Sjors90, 2021-05-28:
I have a anbernic rg351m.
And installed 351 elec firmware.
Everything is working fine until my battery level is under 30-20%.
When I start my system with this battery level, it won’t boot properly.
I see a lot of text and numbers on my screen.
You can see it on the photo below.
But when I charge the system en push the reset button it boots up normally.
I couldn’t find any information about it.
Maybe it is a know issue, I don’t know. I just got my system this week.
I hope you have fix for this 

[image]

Cebion, 2021-05-28:
Normal behaviour on Anbernic Devices.

closed 2021-05-28 by Cebion

Sjors90, 2021-05-28:
Ah thnx 👍🏻