AmberELEC/AmberELEC #442: Wiki says psx supports .zip, but .zip are not found in game scan

alan8r, 2021-05-28:
As the title says, in the supported file formats listed for psx emulation, both .zip and .7z compressed archives are shown as supported, but they don't appear when scanned.

sertiana, 2021-05-28:
> As the title says, in the supported file formats listed for psx emulation, both .zip and .7z compressed archives are shown as supported, but they don't appear when scanned.

.zip is hidden by default. Regardless, it's advised to unzip the game, as the system takes time to extract the game before running it, and there is a possibility that the zipped game will not work at all.

alan8r, 2021-05-28:
Wouldn't it then be better to *not* list .zip as supported by default if it's borderline-disabled without configuration?

sertiana, 2021-05-28:
> Wouldn't it then be better to _not_ list .zip as supported by default if it's borderline-disabled without configuration?

Well, it is technically still supported, with mentioned risks.
Although I agree they should update the wiki page.

konsumschaf, 2021-05-28:
.zip and .7z are disabled - I changed the wiki accordingly. Thanks for pointing this out.