AmberELEC/AmberELEC #465: Can't start Dreamcast games (retrorun or libreto) from 2nd sd slot. Guru meditation

mtoensing, 2021-06-06:
### Describe the bug
I flashed a new image of Crazy Hedgehog (Hotfix3). Put the bios collection on the same card as some dc games. Everything works fine. Now I put the same games and bios files in the 2nd sd card slot. I get a guru meditation on libreto and a "back to menu" in retrorun. 
Every other system (psx, snes, megadrive, nes ...) works fine from the 2nd slot. 
This worked fine on the first 351ELEC reborn release.
Where do I find the logs? 

### To Reproduce
Steps to reproduce the behavior:
1. Flash crazy hog 3 on an sd card.
2. Put bios and dc games on a second sd card (exfat)
3. Start the dc games (shenmue, sc, sa2)
4. See error "guru meditation"

### Expected behavior
Dreamcast games should work from 2nd sd card like all other games do.

### Device (please complete the following information):
Crazy Hog 3

dhwz, 2021-06-06:
logs are in /tmp/logs but I don't think this is a bug

mtoensing, 2021-06-06:
I started with a fresh image to verify this. What could be the problem if this is not a bug? 

I don't see a problem in the log files. Do you? 

[exec.log](https://github.com/351ELEC/351ELEC/files/6604066/exec.log)
[es_log.txt](https://github.com/351ELEC/351ELEC/files/6604067/es_log.txt)
[es_launch_stdout.log](https://github.com/351ELEC/351ELEC/files/6604068/es_launch_stdout.log)

dhwz, 2021-06-06:
can you please change the log level in developer settings to debug, also if you start it with retroarch you need to enable logging to file there first and save it for testing

mtoensing, 2021-06-06:
[es_log (1).txt](https://github.com/351ELEC/351ELEC/files/6604075/es_log.1.txt)

2021-06-06 13:01:01	INFO	 req sound [basic.launch]
2021-06-06 13:01:01	INFO	   (missing)
2021-06-06 13:01:01	DEBUG	ApiSystem::executeEnumerationScript -> batocera-systems --filter dreamcast
2021-06-06 13:01:02	INFO	Attempting to launch game...
2021-06-06 13:01:02	DEBUG	AudioManager::deinit
2021-06-06 13:01:02	INFO	Attempting to close SDL AUDIO
2021-06-06 13:01:02	INFO	SDL AUDIO Deinitialized
2021-06-06 13:01:02	INFO	Configure emulators command : -p1index 0 -p1guid 03000000091200000031000011010000 
2021-06-06 13:01:03	DEBUG	fireEvent: game-start /storage/roms/dreamcast/Soulcalibur\ \(Europe\)\ \(En,Fr,De,Es\).gdi Soulcalibur (Europe) (En,Fr,De,Es)
2021-06-06 13:01:03	INFO		/usr/bin/runemu.sh /storage/roms/dreamcast/Soulcalibur\ \(Europe\)\ \(En,Fr,De,Es\).gdi -Pdreamcast --core=flycast --emulator=libretro --controllers="-p1index 0 -p1guid 03000000091200000031000011010000 "
2021-06-06 13:01:08	WARNING	...launch terminated with nonzero exit code 256!

Same with every other DC game. If I use the card in the first slot it works perfectly. All other systems on the 2nd slot card work without a problem (SegaCD, PSX, etc)

mtoensing, 2021-06-06:
Okay, I will copy all files to the sc again and check if this is an sd card error.

mtoensing, 2021-06-06:
It has something to do with some .chd and some gdi/bin files. Not a problem with the software I thing. Sorry to bother you.

closed 2021-06-06 by mtoensing

mtoensing, 2021-06-06:
Some .chd file for dreamcast don't seem to run on crazy hog. Soul Calibur (USA).chd won't run but Soul Calibur (JAP).chd does. No matter what slot. That was my problem. I started JAP on slot 1 and later USA on slot 2. Weird. I am now trying GDI.