AmberELEC/AmberELEC #466: RetroRun N64 emu doesn't work c-pad

fensjoy, 2021-06-06:
## This is only for actual bugs that occur on the system! For Support visit us in our Discord
https://discord.gg/bmXtCt88Tz

### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.
version firmware 20210603
### Describe the bug
on RetroRun n64 emul doesn't work right stick as a c-pad

### To Reproduce
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

### Expected behavior
A clear and concise description of what you expected to happen.

### Screenshots
If applicable, add screenshots to help explain your problem.

### Device (please complete the following information):
 - Version of 351ELEC that you are using

### Additional context
Add any other context about the problem here.

dhwz, 2021-06-06:
nobody said it does this is not a but it's not supported right now

closed 2021-06-06 by dhwz

fensjoy, 2021-06-06:
ok