AmberELEC/AmberELEC #469: Dreamcast not working after Hotfix 3

fry-guy, 2021-06-07:
I just installed hot fix 3 and Dreamcast simply does not work. I’ve got at 10-15 hours logged on Skies of Arcadia and now the game won't load. Tried other games, same issue. Everything worked fine before the update... very well, in fact. I tried all three core options. Something seems very wrong. Anyone know why this suddenly isn’t working?

dhwz, 2021-06-07:
Please report that upstream at the libretro git then we can't fix that.

closed 2021-06-07 by dhwz

fry-guy, 2021-06-07:
I can report upstream to libretro - but it's not just their core. I'm also encountering the problem with the other Dreamcast cores.

dhwz, 2021-06-07:
What do you mean it's not their core? Which other cores? As have only one flycast.

mtoensing, 2021-06-09:
Ah, this might be the same cause as this one: #465