AmberELEC/AmberELEC #477: Drastic has slowed down significantly vs Reborn release.

Marxiboy, 2021-06-09:
### Describe the bug
Enhanced 3D used to work very well on Reborn release with the very same version of Drastic. Now Drastic is too slow to activate that function, extreme lag

### To Reproduce
Steps to reproduce the behavior:
1. Start any game (Mario Kart DS for test purpose)
2. Go into options - video options - turn on Enhanced 3D
3. Extreme slowdown. Used to work very well with the same Drastic version on Reborn.

### Expected behavior
Work as good as in Reborn.

 - 351M with 351elec Crazy Hedgehog hotfix3

### Additional context
Tried to simply go back to Reborn, but Retrorun Flycast has been removed from downloads, so it's now impossible to have both dreamcast and drastic working properly.

dhwz, 2021-06-09:
Drastic is a precompiled binary we didn't touch it.

Marxiboy, 2021-06-09:
Still, something is breaking its performance, it has been seen around the facebook group. If Drastic, or the thing breaking it cannot be fixed, is there a way to go back to Reborn and install the Retrorun Flycast so I have both a working Drastic and Retrorun Flycast? This was removed from the download section.

Marxiboy, 2021-06-10:
I have new informations. Mario Kart seeems to only lag for a bit when Enhanced 3D is on then run smooth. I'll check further. Maybe preload rom or something like that?

Marxiboy, 2021-06-10:
I said earlier that I thought I had some convincing results. I stopped laggin after a few seconds because I was in first place with very few 3d models in front of me. The performance, even without Enhanced 3D, has dropped by a lot. Is it possible that "Max Performance" is off? As users, we don't have access to that setting. Just throwing ideas and see what sticks.

dhwz, 2021-06-10:
Cause may be libsdl but that can't easily be upgraded as a lot of other packages were broken with the newer version.

Shodopan, 2021-06-10:
Was also able to replicate this issue.

AlexParkerNY, 2021-06-11:
it's true, in the reborn  version you could play even without frame skip and it always had good performance, I hope you can find some solution to that problem.

dhwz, 2021-06-11:
Right now no, as this will break other packages.

Marxiboy, 2021-06-11:
Does anyone still have the package that was at https://github.com/351ELEC/packages/releases/download/1.0/retrorun.zip|8bb1f7b61c7e104fc7b0c50d5b90fe0795cfbb2b85b9d46b633847ad9d1951de
I'm trying to manually merge it with Reborn.

Cebion, 2021-06-11:
> Does anyone still have the package that was at [...]

The drastic problem comes from sdl2.10 we downgrades because some packages had problems.
But some packages like drastic perform a bit slower.

We're working on a fix.

Marxiboy, 2021-06-11:
> > Does anyone still have the package that was at [https://github.com/351ELEC/packages/releases/download/1.0/retrorun.zip|8bb1f7b61c7e104fc7b0c50d5b90fe0795cfbb2b85b9d46b633847ad9d1951de](https://github.com/351ELEC/packages/releases/download/1.0/retrorun.zip%7C8bb1f7b61c7e104fc7b0c50d5b90fe0795cfbb2b85b9d46b633847ad9d1951de)
> > I'm trying to manually merge it with Reborn.
> 
> The drastic problem comes from sdl2.10 we downgrades because some packages had problems.
> But some packages like drastic perform a bit slower.
> 
> We're working on a fix.
> 
> Has nothing to do with retrorun

Thanks for the answer. I'm simply trying to go back to Reborn for now, but Retrorun Flycast was removed from the download section so I will lose that function if I go back. I'm just searching for a workaround.

Tokodomo, 2021-06-11:
Yes, libSDL strikes back again… You can use libSDL2-2.0.so.0 from Reborn release with Drastic like this:
```
mkdir /storage/tmpworkdir

cd /storage/tmpworkdir

wget -qO- https://github.com/351ELEC/351ELEC/releases/download/20210510/351ELEC-RG351P.aarch64-20210510.tar | tar xv

mkdir mnt

mount 351ELEC-RG351P.aarch64-20210510/target/SYSTEM mnt

mkdir /storage/drastic/lib

cp -a mnt/usr/lib/libSDL2-2.0.so.0.12.0 /storage/drastic/lib/libSDL2-2.0.so.0

umount mnt

mv /storage/drastic/drastic.sh /storage/drastic/drastic.sh.bak

nano /storage/drastic/drastic.sh
```
-> Copy and paste into nano:
```
#!/bin/bash

source /etc/profile

BINPATH="/usr/bin"
EXECLOG="/tmp/logs/exec.log"

export LD_LIBRARY_PATH=/storage/drastic/lib

cd /storage/drastic/aarch64/drastic/
maxperf
./drastic "$1" >> $EXECLOG 2>&1
normperf
```
^ Press `Ctrl + O` to save. To exit nano, press `Ctrl + X`. 
```
cd ..

rm -R tmpworkdir
```

Works great for me.

Marxiboy, 2021-06-11:
That would be in SSH yes? I'm going to give this a spin immediately. Huge preach!

Tokodomo, 2021-06-11:
Yes, via SSH.

Shodopan, 2021-06-11:
Will love to give this a try but don't know what SSH and Nano is.

AlexParkerNY, 2021-06-11:
Ssh is the wifi file transfer function, But I don't know what nano is

Shodopan, 2021-06-11:
Would this worh if I just open the drastic.sh file in the sd card with notepad and edit the contents to this?

#!/bin/bash

source /etc/profile

BINPATH="/usr/bin"
EXECLOG="/tmp/logs/exec.log"

export LD_LIBRARY_PATH=/storage/drastic/lib

cd /storage/drastic/aarch64/drastic/
maxperf
./drastic "$1" >> $EXECLOG 2>&1
normperf

Marxiboy, 2021-06-11:
> Yes, libSDL strikes back again… You can use libSDL2-2.0.so.0 from Reborn release with Drastic (in Crazy Hedgehog release) like this:
> 
> ```
> mkdir /storage/tmpworkdir
> 
> cd /storage/tmpworkdir
> 
> wget -qO- https://github.com/351ELEC/351ELEC/releases/download/20210510/351ELEC-RG351P.aarch64-20210510.tar | tar xv
> 
> mkdir mnt
> 
> mount 351ELEC-RG351P.aarch64-20210510/target/SYSTEM mnt
> 
> mkdir /storage/drastic/lib
> 
> cp -a mnt/usr/lib/libSDL2-2.0.so.0.12.0 /storage/drastic/lib/libSDL2-2.0.so.0
> 
> umount mnt
> 
> mv /storage/drastic/drastic.sh /storage/drastic/drastic.sh.bak
> 
> nano /storage/drastic/drastic.sh
> ```
> 
> -> Copy and paste into nano:
> 
> ```
> #!/bin/bash
> 
> source /etc/profile
> 
> BINPATH="/usr/bin"
> EXECLOG="/tmp/logs/exec.log"
> 
> export LD_LIBRARY_PATH=/storage/drastic/lib
> 
> cd /storage/drastic/aarch64/drastic/
> maxperf
> ./drastic "$1" >> $EXECLOG 2>&1
> normperf
> ```
> 
> ^ Press `Ctrl + O` to save. To exit nano, press `Ctrl + X`.
> 
> ```
> cd ..
> 
> rm -R tmpworkdir
> ```
> 
> Works great for me.

This workaround works perfectly! Thank you soo much good fellow.

acridAxid, 2021-06-11:
Let's move all workaround discussion to the Discord and keep this discussion clear for those working on implementing a solution for next 351ELEC update.

closed 2021-06-18 by dhwz