AmberELEC/AmberELEC #479: GBA roms only loading white screen

bpg2210, 2021-06-10:
I recently switched over from the stock OS on my RG351P to 351ELEC and am having issues playing all GBA games. I have tried deleting and re-downloading the bios and the rom files with no change. All other system emulators seem to launch and run smoothly.

sertiana, 2021-06-10:
If your roms are zipped, try to unzip your roms first.

bpg2210, 2021-06-10:
> If your roms are zipped, try to unzip your roms first.
> Second, make sure you are using a genuine SD Card. Bad SD Card may corrupt the files inside it.

Roms are unzipped and using a Samsung SD card.

hahnlo, 2021-06-10:
Does this happen on all GBA games or only the games that you have played on Stock OS? It may have something to do with auto loading an incompatible savestates so try disabling it. Anyway, this is not a place to get help like this, join the discord instead.

bpg2210, 2021-06-10:
It's happening on all GBA games. auto save/load states is turned off. 
also, i tried the 351Elec discord and maybe im missing something but everything just appears to be announcements and the 351Elec Github tab only has links to new comments in this forum

RavenKilit, 2021-06-10:
Read the rules channel. You have to read and react to the rules message, then you'll see all channels