AmberELEC/AmberELEC #480: Drastic emulator slow performance

AlexParkerNY, 2021-06-10:
## This is only for actual detailed feature requests for the system! For Support or if you want to discuss ideas visit us in our Discord
https://discord.gg/bmXtCt88Tz

### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.

### Is your feature request related to a problem? Please describe.
A clear and concise description of what the problem is. Ex. I'm always frustrated when [...]

### Describe the solution you'd like
A clear and concise description of what you want to happen.

### Describe alternatives you've considered
A clear and concise description of any alternative solutions or features you've considered.

### Additional context
Add any other context or screenshots about the feature request here.

dhwz, 2021-06-10:
We don't have a new version

closed 2021-06-10 by dhwz

AlexParkerNY, 2021-06-10:
What could have caused that? I remember upgrading to 351 elec and after that the drastic performance decreased