AmberELEC/AmberELEC #487: PCSX-ReARMed latest update causes controller d-pad not working on RG351V

crksn36, 2021-06-14:
I frequently play PCSX-ReARmed emulator with some selected psx games. the latest update 351elec 20210603 Crazy  Hedgehog hotfix for RG351V messes up dpad controller even I try remapping but doesn't work, every games I load every each doesn't work and I switch to SwanStation emulator which had a little bit laggy in sounds, graphics and movement. The previous release of 351elec 20210510 and 20210422 I had no problems with controllers. I hope can look this matter thanks!

dhwz, 2021-06-14:
reset your device there is no issue with the controls

closed 2021-06-14 by dhwz

crksn36, 2021-06-14:
what do you mean reset? what do I need to do? step by step?

dhwz, 2021-06-14:
Come to Discord for support this, issues is for real issues only.

alien2003, 2022-11-04:
> Come to Discord for support, issues is for real issues only.

Not everyone can sign up for Discord because of their draconic verification system. For example, I can't sign up at all, they ask me to verify my phone number, then ask me to verify my email and permaban me after that. Tried with multiple different phone numbers