AmberELEC/AmberELEC #490: Image won't load, device stuck with balack screen

viniciusccarvalho, 2021-06-15:
## This is only for actual bugs that can be reproduced on multiple systems!  ##
 For Support visit us in our Discord
https://discord.gg/bmXtCt88Tz

### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.

### Describe the bug
Tried to install on device and after the multiple 351 logos appear it goes into a black screen and get stuck 

### To Reproduce
Formatted 16gb SD card using disk utils
Loaded the image using Apple Pi Baker 2
After loading my sd was formatted as follows:
```
/dev/disk3 (external, physical):
   #:                       TYPE NAME                    SIZE       IDENTIFIER
   0:     FDisk_partition_scheme                        *15.5 GB    disk3
   1:             Windows_FAT_32 351ELEC                 1.1 GB     disk3s1
   2:                      Linux                         33.6 MB    disk3s2
                    (free space)                         14.4 GB    -
```

Put it on my RG351V and it hangs on a black screen

### Expected behavior
Install to succeed

### Device (please complete the following information):
 - Version of 351ELEC that you are using : 351ELEC - 20210603 Crazy Hedgehog (Hotfix3)

dhwz, 2021-06-15:
Did you flash the right image for the V?
For support come to Discord.

closed 2021-06-15 by dhwz

Cebion, 2021-06-15:
You have to accept the rules in discord. Reproducible means other people can recreate your issue.