AmberELEC/AmberELEC #498: Menus missing in PPSSPP

sebvalmontx360, 2021-06-19:
### Describe the bug
Menus in PPSSPP are missing. I can see the previously accessed games but the menus do not appear. 

### To Reproduce
Steps to reproduce the behavior:
1. Go to Tools
2. Click on start PPSSPP
4. See error

### Expected behavior
I should see PPSSPP menus, to change settings, etc. 

### Screenshots
If applicable, add screenshots to help explain your problem.

### Device (please complete the following information):
Rg351v with 351ELEC Crazy Hedhog

### Additional context
Add any other context about the problem here.

acridAxid, 2021-06-20:
This sounds like the PPSSPP folders and files have gone missing on the users second SD card.  Those files are placed there when the firmware first initializes, but won't be there if the user re-formats their card or wants to use multiple SD cards for 

Is there a reason why those files aren't stored on the read-only filesystem and symlinked on demand as needed (config file modified at execute, probably)?

It might be a good idea to store the initial state of these files on the read-only filesystem anyway and then add a function to the 'Danger Zone' (something like 'initialize storage partition') so users who may have accidentally blown away their 'ppsspp' folders or want to set up a second SD card after firmware initialize won't need to understand these problems and manually drag the files along with them?

sebvalmontx360, 2021-06-20:
I am using two SD cards for games. 

How can I restore such files to correct this problem?

sebvalmontx360, 2021-06-20:
Thank you. I found a way to solve... I flashed another card and copied the gamedata folder to my two game SDs.

closed 2021-09-16 by dhwz