AmberELEC/AmberELEC #499: N64 control bindings dont work

pooooutine, 2021-06-20:
## This is only for actual bugs that occur on the system! For Support visit us in our Discord
https://discord.gg/bmXtCt88Tz

### Describe the bug
dpad controls on n64 don't work without configuring. to correct for this, controls need to be redefined from the preset of "up, down, left, right" to "c-buttons y+, c-buttons y-, c-buttons x+, c-buttons x-"

### To Reproduce
Steps to reproduce the behavior:
1. open any n64 game
2. dpad controls dont work

### Expected behavior
dpad to function like dpad

### Device (please complete the following information):
 - RG351V

Cebion, 2021-06-20:
This is no support forum. Visit discord for support

dhwz, 2021-06-20:
And how do you use the real dpad buttons then?
The c buttons are already mapped to r2+abxy

masterdev72, 2021-06-25:
I too am having problems with this bug. (I call it a bug for now because I can't find a config solution yet). 

If it helps; this is my take. 

I've updated my 351p by coping .tar file to update folder and rebooted. Since update retroarch doesn't seem to map Control sticks X and Y to left analog stick; if I use C Buttons X and Y i can use these to map to the left analog but this setup won't allow me to use right analog for my c buttons. The bug comes down to the four Control Stick button settings aren't usable for mapping in any N64 core. 

I haven't tried reflashing my SD card yet as on the 351p it is a long process due to the single SD card slot forcing a games copy off then back. 

Hope this makes sense and helps.

GorillaWarHead, 2021-11-20:
Also the R1 button doesn't work as n64 R button and c buttons are mapped to r1 + abxy and not r2 + abxy

dhwz, 2021-11-20:
@GorillaWarHead yes thats the only valid point in this issue thread
Everything else written above makes no sense and/or is related to retroarch itself and has nothing to do with our image.

navy1978, 2021-11-21:
Fixed, in retrorun now buttons L1,L2,R1and R2 are aligned to retroarch, please merge 351Elec retrorun  with mine, commit: 9214474b3be936c7576dc267e215fdc59e9808af

dhwz, 2021-11-21:
@GorillaWarHead are you available for a test? If yes please contact me on Discord.

GorillaWarHead, 2021-11-21:
> 

i contacted you on Discord

closed 2021-11-21 by dhwz