AmberELEC/AmberELEC #504: Vídeo missing in Retrorun Parallel 64

sebvalmontx360, 2021-06-21:
## This is only for actual bugs that occur on the system! For Support visit us in our Discord
https://discord.gg/bmXtCt88Tz

### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.

### Describe the bug
I dont have any vídeo using retrorun Parallel 64 core. I can hear the audio, but there is no vídeo. The problem does not occur with retroarch cores and retrorun parallel glde. I tried with F zero, Dr Mario 64 1080 snowboardibg with same results

### To Reproduce
Steps to reproduce the behavior:
1. Select retrorun parallel 64 as the default N64 core. 
2. Open any N64 game. 

### Expected behavior
I was expecting tô see vídeos, like in the other cores. 

### Screenshots
If applicable, add screenshots to help explain your problem.

### Device (please complete the following information):
 - Version of 351ELEC that you are using
 - Rg351v with Crazy Hedhog

### Additional context
Add any other context about the problem here.

Sephzer, 2021-06-21:
Can also confirm that I have the same issue with the retrorun Parallel 64 core on the RG351v with Crazy Hedgehog. This is the same for Super Mario 64, Paper Mario and Zelda Ocarina of Time (and basically anything else in my library). It just quits back to the menu.

dhwz, 2021-06-21:
@Sephzer
You didn't unzip your roms, .zip/.7z are not supported check our wiki.

Sephzer, 2021-06-21:
Ok so that fixed the game loading issue (thanks, should have done some reading there, though I couldn't find much in the logs), However, this bug is still accurate, there's only sound no video.

sebvalmontx360, 2021-06-21:
My roms are not zipped. They are z64, v64 and N64 files

sebvalmontx360, 2021-06-21:
Just to explain, I had problems with PPSSPP menus missing... however, I restored my gamedata folder from a clean installation and it worked.

Should I restore another file/folder for retrorun -> Parallel64

dhwz, 2021-07-07:
fixed on next release

closed 2021-07-07 by dhwz