AmberELEC/AmberELEC #515: opensim hardware osh pb controller disconnected and reconnects

mwalton07, 2021-06-26:
351 elec opensim hardware osh pb controller disconnected and reconnects after the crazy hedgehog update while using a TP Link TL-WN725N wifi adapter in the left otg usb c port. It disconnects and reconnects for a min then it stops and 15 to 20 Mins later it does the same disconnecting and reconnecting.

[image]
[image]

TheOuterLinux, 2021-06-29:
I've noticed a similar issue for when the battery drops somewhere between 30-40%.

mwalton07, 2021-06-29:
I have tried the wifi dongle in the right OTG port and it's fine. It seems to do this in the left OTG port

pkegg, 2021-07-20:
Unfortunately, I don't think there's much we can do here.  

The message you're seeing is tied directly to the Linux kernel (re)detection of the USB controller.  I would guess there's some kind of interference going on at a hardware level causing the detection/redection of the controller when you have the wifi card in the left port (or maybe there's a drop in electrical current, etc).

Whatever is happening, at the 351ELEC software level, we just get the messages that the controller is (re)detected.  There's not much we can do.

Given that we cannot really fix this and you have a workaround (use the right OTG port), I'm going to close this issue.  Feel free to post on discord if you are in need of further support.  Thanks for your time and responses.

closed 2021-07-20 by pkegg