AmberELEC/AmberELEC #520: Use lzdoom

KrumpetPirate, 2021-06-29:
## This is only for actual detailed feature requests for the system! For Support or if you want to discuss ideas visit us in our Discord
https://discord.gg/bmXtCt88Tz

### Getting started
Before opening an issue, check the release notes of the version that you are using and recent commits.  Be sure to fill out the template completely, accurately, and with as much detail as possible.  Issues without sufficient detail will be rejected and closed.

### Is your feature request related to a problem? Please describe.
Doom mod performance is very poor with whatever zdoom port was used here. ArkOS uses lzdoom to good result and I think the doom play on 351elec should follow that example

### Describe the solution you'd like
Switch doom engine from whatever zdoom is being used (gzdoom or zdoom) to lzdoom instead

### Describe alternatives you've considered
No other alternatives were considered as it seems that lzdoom gives the best performance

### Additional context
Playing Brutal Doom Lite on each firmware gives clear indication that ArkOS's doom emulation is much more performant.

KrumpetPirate, 2021-06-29:
Looking at the code it would appear that lzdoom is being used. So why is there such a big performance difference between the firmwares?

dhwz, 2021-06-29:
This is not a competition, don't compare systems...

closed 2021-06-29 by dhwz

KrumpetPirate, 2021-06-29:
No it certainly isn't, it's open source software. We can take pros and cons for each firmware to move the system forward. I'll look into it but I doubt I'll be able to make much progress. I meant no offense.